quartiq/phaser #1: DSP interpolation path specs

jordens, 2019-10-29:
Currently the plan is to go for 25 MS/s IQ per channel and 20 MHz usable bandwidth. Then the interpolation cascade is 1/10 (samples per 4 ns FPGA clock cycle) -> 1/5 -> 2/5 -> 2/1.
Now we should determine a few things on that interpolation filter to maximize flexibility and minimize power/resources/compile time.

1. Pass band flatness: Assume the passband is -10 to 10 MHz. If e.g. 1e-1 (-0.9 dB) amplitude droop at the base band edges is acceptable, then we can skip the CIC compensation filter. If not, then 1e-2 or 1e-3 or 1e-4 in amplitude flatness? Contrast this with AOM/Booster response flatness.
2. Stop band attenuation: 70 dB would permit a simplified architecture of the first and biggest AA filter. Otherwise 80 or 90 dB? Contrast this with how much power you can accept in Nyquist images outside the +-10 MHz bandwidth and the damage these images can cause there. Also note that If your baseband signal (short pulses) contains power in the transition band (from -12.5 to 10 and from 10 to 12.5 MHz), then that's only mildly attenuated (between 0 dB and 6 dB) and will always leak to the respective other side of the band.
3. Is it worth going from 0.4 (+-10 MHz of 25 MS/s) usable bandwidth to 0.42 (+-10.5 MHz)? The cost is about 15 dB in stopband attenuation for a given filter complexity, c.f. (2).

The four tone generators are pretty much clear. For the architecture we are looking at 105 dBc SFDR, 32 bit FTW, truncation to 19 bit phase word, 16 or 14 bit outputs.

@hartytp

hartytp, 2019-10-29:
> Pass band flatness: Assume the passband is -10 to 10 MHz. If e.g. 1e-1 (-0.9 dB) amplitude droop at the base band edges is acceptable, then we can skip the CIC compensation filter. If not, then 1e-2 or 1e-3 or 1e-4 in amplitude flatness? Contrast this with AOM/Booster response flatness.

A 10% amplitude change over the +-10MHz has the potential to be a little annoying in some situations. 1% would be completely fine. Whether or not it's actually worth it depends on the complexity/time tradeoffs, which I find hard to judge.

> Stop band attenuation: 70 dB would permit a simplified architecture of the first and biggest AA filter. Otherwise 80 or 90 dB? Alternatively something like 40 dB attenuation and 6 dB droop would allow getting rid of all FIR filters and just using a single CIC interpolator. Contrast this with how much power you can accept in Nyquist images outside the +-10 MHz bandwidth and the damage these images can cause there.

How does stop-band attenuation translate into SFDR? It would be nice to keep the SFDR to >60dBc or so, 70dBc feels quite comfortable (and, anyway, beyond that the DAC itself would become the limitation AFAICT).

> Is it worth going from 0.4 (+-10 MHz of 25 MS/s) usable bandwidth to 0.42 (+-10.5 MHz)? The cost is about 15 dB in stopband attenuation for a given filter complexity, c.f. (2).

This is not something I'm particularly sensitive to right now, so I'm happy either way. I assume this is just filter coefficients, so something that can be changed relatively easily later on if one wants to really optimize for a specific application?

cc @cjbe

hartytp, 2019-10-29:
> Also note that any power from your baseband signal (short pulses) in the transition bands (from +-10 MHz to Nyquist (+-12.5) respectively) is only mildly attenuated (between 0 dB and 6 dB) and will always leak to the respective other transition band.

You mean spectral content at f0 aliasing to -f0?

jordens, 2019-10-29:
Stop band attenuation means the attenuation of aliases/images of your baseband signal outside baseband. There are aliases every 25 MHz. Without aliasing filter they are at 0 dBc.

Yes. The coefficients are free. But (a) they are integers with a limited range (b) there is a limited number and spacing of filter taps and (c) they depend on each other.

Then let's go for 1e-2 passband amplitude flatness (0.1 dB), 70 dB stop band attenuation, and 0.4 usable width (+-10 MHz) for now:

[image]

Blue are the limits as specified, orange is attenuation (response), green is flatness (1-response).

jordens, 2020-09-09:
Implemented in 11219ee (improved passband ripple and wide(r) band attenuation a lot since there were unused cycles).

[image]

Notebook as of https://nbviewer.jupyter.org/github/quartiq/phaser/blob/ad26c95/filter.ipynb
See the impulse response at the end, individual tap coefficients a couple cells before that.

commit ad26c95b (2020-09-09): filter: finalize, match implementation

close #1
closed 2020-09-09 by jordens