quartiq/phaser #2: STFT processor

jordens, 2020-02-07:
Proposal for a different way to generate "pulses with multiple tones" and equivalently "pulse shaping".

## Background

In virtually all current proposals or experiments on gates (e.g. [LUH](http://de.arxiv.org/pdf/1911.03954.pdf), [NIST](https://arxiv.org/pdf/1910.14178.pdf), [Sussex](https://arxiv.org/pdf/1805.07351.pdf), [Weizmann](https://arxiv.org/pdf/1911.03073.pdf) and others) use multiple tones.

* It's clear from the Physics that the gate is best described and parametrized in frequency space: The phase space trajectory is just the (short-time) Fourier transform of the gate pulse. Probably and hopefully a well known fact but in a quick search I didn't find that idea anywhere. The accumulated phase is the area under the STFT.
* Sometimes the rationale for these multiple tones is a bit hidden and arises from practical arguments on distortion/nonlinearities/intermodulation (in to amplifiers, AOMs etc) or spectral narrowness. Then people call it "pulse shaping" (temporal). But sometimes they forget to view it in frequency space: the temporal shape of a gate pulse (in terms of its samples) has a one-to-one correspondence to its spectral shape (the discrete Fourier transform of the samples). Ergo (and in many cases this is actually much easier to understand) temporal pulse shaping can always be represented as (a few) discrete tones. Meeting the goals of (temporal) pulse shaping is in fact easier done in frequency space.
* The tones typically spaced equally. Or can be chosen to be at equal spacing by some other knob. Or in the worst case might be are few (~2) sets of tones at common and equal intra-set spacing but offset to each other.
* The tones are at multiples or submultiples of the gate duration (obviously by virtue of the temporal-spectral correspondence). This also implies equal spacing.
* The tones are offset to some arbitrary carrier frequency (e.g. "bare qubit").
* Sometimes there are additional requirements arising from technical imperfections. E.g. a drifting carrier during the gate.

## Layout

Drawing the consequences it appears that a very efficient parametrization of a gate pulse would be just:

1. The complex tone amplitudes. `a_m = i_m + j*q_m`, `m` tone index, integer `\in [-M/2*cutoff, M/2*cutoff]`, `M=64, 256, or 1024` FFT width, `cutoff=0.8` interpolation passband
2. The tone spacing. `f_delta = 1/(M*r*t_sample)`. `t_sample = 2 ns` (for Phaser), `r \in [1, 2, 4, 8, 12, ..., infty]` interpolation factor.
3. The duration of the gate. `t_pulse = n*r*t_sample`,  `n` integer pulse length, multiples of `n = M/2` are interesting values.
4. The carrier frequency. `f_carrier` with mHz resolution.

The output tones are then at `m*f_delta + f_carrier` with amplitude `a_m` times the shape of the frequency-shifted gate (square t, i.e. sinc f) times the carrier interpolation window (the typical steep lowpass) shape. No other magic required.

You can skip/clear as many of the `M*cutoff` tones as you want (This is obviously unlike time domain samples where you can't skip). The above would completely describe a pulse. There would be no need for any other data or features ("spline interpolation" or any additional interpolation, "amplitude modulation", "temporal pulse shaping", "pre-distortion").

Technically speaking: you want to take the tone amplitudes, do an FFT, gate the output, interpolate to the pulse length and upconvert to the carrier. The block diagram of the path is simply: FFT, time gate, interpolator, upconverter. No point in sketching it.

With an FFT in FPGA you can easily get a very respectable number of tones out of this.

And you could have more of those chains added together before the DAC.

## Analysis

For Phaser, this would give much more than the "4 tones in 20 MHz" limitation and is much more scalable.

Some IMD and non-linearities can and should be addressed outside this proposal. E.g. chirps and IMD should see something like an extended RFPAL approach (https://github.com/sinara-hw/meta/issues/50#issuecomment-582374158).

This may be the ultimate toolkit do do gates and it may actually make the entire design significantly simpler (on top of making it much more scalable and better than e.g. a fixed set of tones like SAWG or old-Phaser).

With only five tones for example you can do a gate at some given detuning of your choice that is compensated to second order for detuning fluctuations and is minimal peak power or total energy on top of that.

You also get free compensating of non-flat frequency responses out of this. It doesn't need added functionality.

You can also use this to null away spectral leakage (due to the interpolation and gate window) at multiple frequencies.

/cc @hartytp @dtcallcock @dhslichter @chanlists and whoever is interested.

cjbe, 2020-02-10:
I am not sure one can get the required bandwidth with only a 1024 point FFT.
An example to check my understanding:

I want to do a MS-type gate with motional sidebands at f_c +- 5 MHz, with a gate duration of 1 ms.
Putting the sampling rate high enough for the sidebands to be a little within the 0.8 band cutoff -> full bandwidth 2 * 5 MHz / 0.7 = 14.3 MHz
I thus have 14.3 MHz * 1ms = 14e3 components in my gate Fourier series.
(of these, the non-zero components are clustered in a band of ~20 kHz around +-5 MHz).

Q1) Is this many points in the FFT possible? Or is the expectation that one would need 2 STFT processors, one for the upper and one for the lower sideband (in this example).
(this problem gets worse if one wants to include the energy at the second motional sidebands as well as the first)

Q2) There is a reasonable amount of data to be transferred into the STFT engine (2 sidebands * 20 points * 2 coefficients = 160 words that are non-zero) - how do you foresee this loading being implemented so as not to be a bottleneck?
(cf less data with a more complex parametrisation, of pulse bandwidth, sideband frequencies, and spline pulse envelope / phase modulation).

jordens, 2020-02-10:
Q1: If you want to cover `M*f_delta = 1/(r*t_sample) = 25 MHz` (20 MHz inside cutoff) with `M = 256` points ("components"), you'd go for e.g. `r = 20`. Then choose `n = 25000` for `t_pulse = n*r*t_sample = 1 ms`. This gives `f_delta = 25 MHz / 256 = 98 kHz`. Or different `n` and `r` if you don't like `f_delta`.
The size of the FFT can be small if you can live with the `f_delta` (its resolution and lower limit given a bandwidth) and `t_pulse` (resolution) constraints. That's the question.

Q2: In the worst case you have to transfer those 160 words every pulse (1 ms). That's ok. And what do you mean by 20 points? Also what do you compare this with? A 4-tone pulse in the old architecture (not even shaped in ampltiude) has 4 tones times (2 words for frequency, 1 for amplitude, one for phase) = 16 words. In an STFT it has (4 components times 2 quadratures) plus 2 words for carrier, one for duration, and one for interpolation = 12 words.

jordens, 2020-02-10:
* If you need `f_delta` to be a multiple of the inverse pulse time then you need `n` to be a multiple of `M`.
* If you absolutely need a pure (first order) Hann shape of your pulse (e.g. the LUH paper) then you need `n = M`. If you can live with higher order Hann windows, then some other multiple of `M` will work.

jordens, 2020-02-10:
But I see that due to simple coupling strength considerations one would want multiple (blue and red per mode) clearly separate spectral windows, i.e. either multiple carriers or even multiple independent STFT processors. That would allow small `n/r` (towards `M/n >= 1` integer) even for long pulses and high mode frequencies.

dhslichter, 2020-03-06:
@jordens I haven't had much time to think about this these past weeks, but I would say it sounds quite interesting.  In terms of generating the tones, what kind of distortion/noise floor can one reasonably expect from the FFT with (I am assuming CIC) interpolation?  As long as it is below what the DAC can manage we are fine, but I don't have a sense for how this scales with e.g. number of tones, `r`, and so forth.  

Didi has for a long time wanted to express transport waveforms as Fourier series rather than as splines (as you know!), so it seems this would allow that to be tried out as well.  We would set `f_carrier=0` and just use this at baseband.  As you also point out, this would be a big win for tweaking up predistortion given some non-flat frequency response of amps/filters/etc (be it lowpass, notch, whatever) because you just play with complex Fourier amplitudes, no re-calculation of predistorted waveforms necessary.  

For microwave gates, we have sideband Rabi frequencies in the 1-2 kHz regime, so we need to adjust our sideband detunings at the ~ 10s of Hz resolution level to get good loop closure for really high fidelity gates.  For this, given the ~10-15 MHz splitting between red and blue sideband frequencies, we would need to have two separate small spectral windows (keeping `f_bw` small), one near the RSB and one near the BSB, so that `M` can be reasonable.  In reality, power amplifier nonlinearity means that we currently generate the RSB and BSB in separate DACs and sum them after amplification, but in the longer haul we are moving to lower microwave powers and thus would ideally generate RSB and BSB in the same DAC.  We (and others) also need the ability to add a carrier for dynamical decoupling and carrier manipulations, (potentially amplitude-modulated, phase modulated, and/or frequency-chirpable), which would be yet a third spectral window.  None of this seems like a deal-breaker.  

A couple of questions for now:

1. I'm a little unclear on how a frequency chirp would work in this architecture.  Would it basically be a series of discrete frequency jumps (FFT parameter updates), where you ensure that the time between jumps is much smaller than the size of the frequency jump?  This would work for slow chirps but would fail for faster chirps.  Is there another way to think of doing this?  

2. Our gates currently involve a shaped rise/fall time with a long flat-top in between (most gates are this, not the single Hann of LUH).  How does one handle that here?  The exact shape and duration of the rise/fall are important for achieving maximum gate fidelity, so we would need to have some tuning knobs available.  Rise/fall time durations need to be tuned at the <100 ns level currently.

jordens, 2020-03-08:
* Distortion form an FFT is somewhat worse than optimum (quantization limit, see the TI TMS320 or Xilinx docs for example). It doesn't change significantly with the transform width. But with decent choices on the algorithm and bit widths this can be chosen to be better than the other distoring factors in the chain. Better than 70 dB SNR would be a very realistic target. The DAC distortion is likely not the right benchmark.
* The interpolation could be FIR+CIC as usual and doesn't mean distortion (more than the quantization limit). It's also linear phase. The leakage from the other spectral windows can be dialed down to be much below the quantization.
* For constant-speed transport in strictly linear traps with strictly periodic electrodes a Fourier decomposition makes sense. As you and Didi know I thought about this a long time ago. But once you are outside that regime, I still don't see why a parametrization using a Fourier series is better (what's the metric?) than the alternatives (well-defined time-domain interpolation accounted for in the waveform generation). Take a set of waveforms and look at it in time domain: is it periodic? Look at their spectrum: does it have structure? From what I have seen the answer to both is generally no. Maybe you can implement time-variable interpolation after the transform but that only enables variable-speed transport (not even variable electrode size) and it breaks down once you demand pre-distortion. If there was any progress since or of anybody has thought about it in more depth, then they should speak up.
* For multiple spectral windows it would be useful to investigate if that can be factorized (carrier times red/blue/carrier times identical narrow band pulses). And how the limitations can be handled.
* Re frequency chirp: free yourself from thinking that FFT means "constant tones". A chirped pulse can be just as well expressed in Fourier domain as in time domain. There is a one-to-one correspondence between the two representations. No need to change them mid-pulse. But there is also the option to chirp the upconverter NCO (especially if `n > M*r`). That gets more convenient still if the factorization above works.
* I doubt that any of the current gates I mentioned above use a "shaped rise/fall" and a flat top. You may want to take a look at the pulses from your own paper in time domain! And I'm unaware of a meaningful analysis or derivation of the desired/required pulse shape in any of the "old" experiments with flat top pulses. It's always only a qualitative argument about distortion and spectral width. And again: free yourself from being torn apart by trying to do things that you don't need to do. Pulse fidelity is important throughout the pulse. Start and end are not special. And temporal fidelity corresponds to spectral fidelity. There is no contradiction. Demanding some "exact shape and duration of the rise/fall time" or "< 100 ns rise/fall duration tuning" means that you don't yet represent your pulses in frequency domain. This scheme is not for you then. ;)

dhslichter, 2020-03-09:
> Better than 70 dB SNR would be a very realistic target. 

Ack.

> The DAC distortion is likely not the right benchmark.

So what would be?  It depends on your application, and the type of distortion.  If it's broadband noise that you end up with, that's different from harmonic distortion which is (probably) less critical as long as one is careful about where the `f_carrier` is picked.  

> For constant-speed transport in strictly linear traps with strictly periodic electrodes a Fourier decomposition makes sense.  As you and Didi know I thought about this a long time ago. But once you [...]

I am not pushing hard for Fourier decomposition being the best/right thing to do, but it *is* a different way to represent an arbitrary signal over a finite period of time.  The advantage (why it would be "better") would be relative ease in adjusting parameters to compensate for non-flat frequency characteristics of (DAC, amplifier, trap filters) between the digital waveform and the ion.  Are the waveforms periodic?  In general, no.  And I have not spent time thinking about this in detail, or doing calculations (transport is on the far back burner for me these days); this was merely a point/question since this STFT parameterization would enable one to *try out* the idea for transport.  

> For multiple spectral windows it would be useful to investigate if that can be factorized

Roughly speaking, the answer is "yes", but there would probably be the need for an overall amplitude scaling and time offset between red/blue sidebands to make sure the amplitudes and phases are matched at the ion, due to amplifier gain mismatch/frequency-dependent line loss/channel-dependent time delay.  If these are not matched it degrades the quality of the gates, seen experimentally.  

> A chirped pulse can be just as well expressed in Fourier domain as in time domain.

Yes, thanks :)  Not thinking freely enough...

> But there is also the option to chirp the upconverter NCO

That would also be a nice option, especially for multitone gates, if one is compensating for a drifting secular frequency -- moves all sideband detunings uniformly.  

> I doubt that any of the current gates I mentioned above use a "shaped rise/fall" and a flat top.

The [NIST](https://arxiv.org/pdf/1910.14178.pdf) ones do :)  

> And I'm unaware of a meaningful analysis or derivation of the desired/required pulse shape in any of the "old" experiments with flat top pulses. It's always only a qualitative argument about [...]

[NIST/LLNL theory paper here (open access)](https://iopscience.iop.org/article/10.1088/1367-2630/ab0be5) explains exactly why we care about pulse ramps on and off for the microwaves -- it boils down to being able to account properly for off-resonant carrier effects on the gate, which matter much more for microwave gates than for laser gates in general.  In the language of the paper, we describe this as ensuring that the transform between the lab frame and the "bichromatic interaction picture" where we analyze the gate (an approach borrowed from Christian Roos's work on Ca+ laser gates) approaches the identity.  In practice, in the experiment we implement a 5 us Blackman rise/fall shape with a flat top (~50-100 us) in between.  The specific shape doesn't matter enormously (i.e. nothing magical about Blackman), but the whole thing must meet the criterion that the value in Eq. (21) is the identity.  Experimentally, we can see that small (<100 ns) tweaks of the rise and fall times lead to measurable changes in the final Bell state fidelity at the few tenths of a percent level (a level that we care about now).  In other words, it is not enough just to do a smooth ramp off and on, we are rather sensitive to the exact timing if the ramp is not super slow.  Although I haven't done the math explicitly, it must be equivalent to making sure there is a minimum in the frequency spectrum at the carrier frequency; in other words, we are trying to put a sharp "null" at the qubit frequency in the frequency-domain representation of the BSB + RSB pulses, rather than just narrowing their spectrum more (to avoid the qubit frequency) by slowing the rise and fall times.  One can do this kind of "null" with a square pulse too; with a shaped rise/fall, you're basically making miscalibrations around the "null" in frequency space more forgiving.  

> This scheme is not for you then. ;)

It seems so! :)  At least for some applications.  As long as the "several tones with spline-defined amplitude/phase/freq" parameterization continues to be available, that is probably a better match for us at present.  I was unclear whether this STFT parameterization was intended as a substitute or a complement.

jordens, 2020-03-09:
> So what would be? It depends on your application, and the type of distortion. If it's broadband noise that you end up with, that's different from harmonic distortion which is (probably) less [...]

Exactly. Distortion is neither the whole story nor does it originate from DAC only. There is the upconverter, and the entire rest of the RF chain as well. At least in the digital domain, "distortion" is typically a set of input-signal-dependent spurs and not uncorrelated noise.

> this was merely a point/question since this STFT parameterization would enable one to _try out_ the idea for transport.

Yes. But you don't need the phaser gateware to try expressing transport waveforms in Fourier series. Just do it on your computer first and see whether it makes sense.
If the only reason to use a Fourier series is to do transfer function engineering, then this doesn't really help, right? You'd have to apply the distortion to the (frequency) samples the same way you'd have to apply it to the temporal samples. The only difference is that it might be O(1) (equivalent filter taps) fewer multiplications in frequency space.

> Roughly speaking, the answer is "yes", but there would probably be the need for an overall amplitude scaling and time offset between red/blue sidebands to make sure the amplitudes and phases are [...]

Those would all be still within the factorization. Compensating for narrowband transfer function differences would not be. I.e. if there is something that breaks the hierarchy carrier-sidebands-narrowband then that becomes tricky.

> > But there is also the option to chirp the upconverter NCO
> 
> That would also be a nice option, especially for multitone gates, if one is compensating for a drifting secular frequency -- moves all sideband detunings uniformly.

Absolutely. The hierarchy is universal and mapping it is a good idea.

> > I doubt that any of the current gates I mentioned above use a "shaped rise/fall" and a flat top.
> 
> The [NIST](https://arxiv.org/pdf/1910.14178.pdf) ones do :)

I'm very surprised. From what I can see in fig 1 or the preprint (already from (b) and knowing the trajectory-STFT correspondence), this pulse is not flat top at all. Do you have a time domain plot of that pulse to indulge me? I'd actually bet a couple beer on it not being flat top. The presence of multiple tones (that coincidentally do not correspond to phase modulation) contradicts flat top.

> The specific shape doesn't matter enormously (i.e. nothing magical about Blackman), but the whole thing must meet the criterion that the value in Eq. (21) is the identity. Experimentally, we can see [...]

Yes! Why do you look (in the published paper) at your pulse shapes `f(t)` (for the gradient) and `g(t)` (for the carrier) in time domain? Why don't you express it as a Fourier series (over the gate duration)? Everything else in your analysis is just sines and cosines (in time domain) and just begs to be treated in frequency space! Then you'll immediately see what you want. And you also immediately see why Blackman is ok but not special. There are much more special (and shorter) series expansions in there.

> Although I haven't done the math explicitly, it must be equivalent to making sure there is a minimum in the frequency spectrum at the carrier frequency; in other words, we are trying to put a sharp [...]

Yes yes yes. Do the math in frequency space!

> One can do this kind of "null" with a square pulse too; with a shaped rise/fall, you're basically making miscalibrations around the "null" in frequency space more forgiving.

Yes! I know. Why on earth then stick with `f(t)`/`g(t)` and not `f^(n/t_f)`/`g^(n/t_f)`?

This is exactly why I came up with this idea. The condition of phase space loop closure for a (any) mode at delta is exactly `STFT[pulse_shape](t_gate, delta_mode) == 0` in useful variables. (The gate phase is the area enclosed by the `STFT[pulse_shape](., delta_mode)` closed path and all functional analytic tools to play with that are at your disposal). This is why you want to represent your pulses as an STFT: because the STFT is linear, you can enforce these constraints naturally, and because the STFT is invertible, you can generate the pulses in hardware!

> It seems so! :) At least for some applications. As long as the "several tones with spline-defined amplitude/phase/freq" parameterization continues to be available, that is probably a better match [...]

No! I still think exactly for the pulses in your paper this is just the right thing. You just have to let go of thinking in time domain and temporal pulse shaping.

dhslichter, 2020-03-09:
tl;dr - I like this idea, I want to understand more about how to make it work, when I have some time (goodness knows when) I will play with some Fourier transforms and pulses on my computer to figure out more :)

> Exactly. Distortion is neither the whole story nor does it originate from DAC only. There is the upconverter, and the entire rest of the RF chain as well. At least in the digital domain, "distortion" is typically a set of input-signal-dependent spurs and not uncorrelated noise.

I am generally more worried about close-in distortion (e.g. non-harmonic spurs or elevated noise floor) than harmonic distortion; the analog parts of the chain tend to give more of the latter than the former, the exception being IMD from amps and from the upconverter (as well as feedthrough from imperfect mixer balancing).  We jump through various hoops to try to keep IMD to a minimum (e.g. parallel amplifier chains, summed afterwards on hybrids), and we use homodyne mixing to avoid issues with low-level carrier and sideband feedthrough that one gets with SSB mixing, even with careful tweaking of offset bias.  

Anyway, the DAC will probably be the eventual limit on distortion (we currently try hard to make it so for close-in distortion), but I wanted to get a sense of where this STFT method might have its own noise floor.  Whatever spurs arise need to be controlled carefully so they don't end up in the wrong place (this seems like a tractable issue, but one that we need to at least be aware of).  

> If the only reason to use a Fourier series is to do transfer function engineering, then this doesn't really help, right? You'd have to apply the distortion to the (frequency) samples the same way you'd have to apply it to the temporal samples.

True.  However, if you are already building an engine to turn frequency samples into time-domain output waveforms, then these frequency amplitudes can be tweaked and the rest of the math done in gateware, while if one changes predistortion on temporal samples one has to recalculate -- which in the absence of gateware means math on the PC (or possibly on a Zynq core device), thus slower.  I agree, there is no free lunch, but I was just trying to harness the power of something you're already planning on doing....

> I'm very surprised. From what I can see in fig 1 or the preprint (already from (b) and knowing the trajectory-STFT correspondence), this pulse is not flat top at all. Do you have a time domain plot [...]

Thanks for the beers :)  You think I don't know what's in my own paper?  Read the bottom right corner of p.3.  It is three flat top pulses (in the lab frame), at different frequencies (`w0 + delta`, `w0-delta`, and `wg ~ 2*delta`), each with ramped rise and fall.  The microwave pulses are turned on first, and then the gradient pulse is ramped on once the microwaves have reached steady state amplitude (reverse order at end of the pulse).  In the bichromatic interaction picture it looks like a two-tone ZZ interaction.  This near-motional oscillating gradient scheme allows one to make two effective sideband pairs in this interaction picture using only these three fields in the lab frame.  The gates in the paper use two such phase space loops separated by a pi pulse in order to echo static qubit frequency errors.  

> Yes! Why do you look (in the published paper) at your pulse shapes f(t) (for the gradient) and g(t) (for the carrier) in time domain? Why don't you express it as a Fourier series (over the gate duration)?

It's a good thought; the main (stupid) reason is that the knob we have in the lab is time domain pulse shaping with PDQ splines, so we have been thinking in terms of that.  I will go back and think about shaping and coefficients in the frequency domain.  

> No! I still think exactly for the pulses in your paper this is just the right thing. You just have to let go of thinking in time domain and temporal pulse shaping.

I am not giving up on the STFT idea!  I think it's neat and still sorting through things, trying to understand what the limits might be, and where the big wins are.  I haven't thought about this in great detail, as stated, due to my own bandwidth limitations.  

I still don't see a way around wanting a flat-top pulse, though, simply for reasons of gate speed.  When one uses a pulse that is windowed over its whole duration (e.g. the LUH design), it means the entire gate is slower by a factor of ~2 when one is working in the regime where the maximum Rabi frequency is limited by available power (this is always the case for microwave gates), and one of the issues with microwave gates is speed.

jordens, 2020-03-10:
> I am generally more worried about close-in distortion (e.g. non-harmonic spurs or elevated noise floor) than harmonic distortion; the analog parts of the chain tend to give more of the latter than [...]

Whether you look at the third harmonic of a single tone or close-in third order intermodulation doesn't matter much. They have the same origin, same mechanism, and same scaling, only the stimulus is different. I don't see how an analog component can have more harmonic distortion than IMD. Especially since harmonic distortion is typically lowered due to bandwidth limits and filters while IMD is not.
The difference is mostly in the non-harmonic and non-intermodulation distortion ("spurs"): those are more likely to have a digital origin. Harmonic, IMD tend to be analog. Aliasing and bias/leakage can be either.

> We jump through various hoops to try to keep IMD to a minimum (e.g. parallel amplifier chains, summed afterwards on hybrids), and we use homodyne mixing to avoid issues with low-level carrier and sideband feedthrough that one gets with SSB mixing, even with careful tweaking of offset bias.

This has been done since the early days more than a decade ago IIRC.
I'd be interested to know how how what the SNR is since your "shaping" (or lack thereoff) will still see full IMD of a single channel.

> True. However, if you are already building an engine to turn frequency samples into time-domain output waveforms, then these frequency amplitudes can be tweaked and the rest of the math done in [...]

Gateware to do temporal transfer function compensation is likely significantly simpler than gateware to handle the variable-speed tweaking of Fourier series components. Why should there be an advantage? This still seems to be wild but inefficient idea to me.

> Thanks for the beers :) You think I don't know what's in my own paper? Read the bottom right corner of p.3. It is three flat top pulses (in the lab frame), at different frequencies (`w0 + delta`, `w0-delta`, and `wg ~ 2*delta`), each with ramped rise and fall.

I expect that you know your paper very well. That's not what I meant.
But `w0 + delta`, `w0 - delta` together are a fully modulated and not at all flat. If you choose the phase offset correctly, then the pulse already starts and ends with a smooth envelope naturally (unlike a keyed carrier), because it's modulated. A pulse with those tones and proper phase offset has less power/spectral leakage (far away) than the usual dynamical decoupling pulse with a (unmodulated, only keyed) carrier at `w0`. That's what I mean.
Once you have multiple tones, in all likelihood you have modulation and some "shaping" (save the case of PM/FM). And also, somewhat counterintuitively, a pulse consisting of multiple Fourier components in many cases has less spectral leakage (wideband) than a (keyed) pulse with a single Fourier component. This is before and independent of any additional envelope "shaping".

Individually, the tones of a pulse are always "flat top". Even if you shape the pulse, each component in the Fourier expansion is "flat top" and has "steep and well defined rise and fall". With multiple tones in a Fourier series the notion of (temporal) "flat top" is not that meaningful.

By the way, in your case, the pulse (in the bichromatic interaction picture) looks like it might be equivalent to a phase (or frequency?) modulated pulse. The Jacobi-Anger expansion and the Bessel functions are typical tell-tale of FM/PM. `4 \Omega_µ/\delta` smells like it is in fact a modulation index. That picture could also lend itself to simpler mechanisms of generating and easier parametrization.

> I still don't see a way around wanting a flat-top pulse, though, simply for reasons of gate speed. When one uses a pulse that is windowed over its whole duration (e.g. the LUH design), it means the [...]

IIRC dynamical decoupling comes with a slow-down of similar magnitude. Given constant microwave power, is your scheme just as fast as the pure MS gate? I see a relative gate speed of only 0.49 in your paper. That would be worse than a factor of two slow-down in the LUH design.
I don't think a factor of two is necessarily a bad thing. From what I have heard and IIUC the robustness of the LUH design allows them to go higher in speed before they see issues due to memory/thermal (ask them about their pre-heating pulses to warm up the RF chain and trap). Also note that a Hann pulse is just two Fourier bins instead of one, i.e. naturally STFT-able). My guess is that the spectral narrowness of the two tone (Hann) pulse just avoids contributions due to leakage and IMD and those typcially have much nastier and volatile dependencies on thermal/memory effects that are hard to cancel out.

jordens, 2021-01-14:
Taken up by #14

closed 2021-01-14 by jordens