quartiq/phaser #5: Interpolation filter initialization

jordens, 2020-09-16:
There is currently about a 3/4 chance that the filters initialize incorrectly and output noise or high-harmonic signals. Likely the HBF even/odd timing.

jordens, 2020-09-21:
The the bad initialization resulting in distortion has been fixed in f351d15.

sotirova, 2020-10-12:
Have not been able to initialise the filters correctly either. Have tried load.tcl multiple times. Checking out f351d15 and loading the new bitstream didn't help either. (For reference, I am talking about a phaser baseband board that I am trying to get to work for the first time.)

jordens, 2020-10-12:
Stick with the commits you had. Those were good.
Could you disable/force that amplitude/phase check and look at the RF output on a specturm analyzer? I assume you are using the sinara_tester script or the example.py (do disable the dac_sleep at the end) in the repository.

jordens, 2020-10-13:
And you used this [bitstream](https://github.com/quartiq/phaser/releases/download/v0.4/phaser_v0.4.bit)?

sotirova, 2020-10-13:
Thanks for the comments! I am using the example.py script for testing and have disabled the dac_sleep at the end. I compiled the bitstream myself having checked out 269724a454034cf86b9211482bf57cb89e59a0ba. I have now disabled the oscillator test however I am not seeing any RF output.

jordens, 2020-10-13:
Could you try the bitstream I linked? Just load it after kasli has booted. Then run the example.

sotirova, 2020-10-13:
Just tried - still unhappy (i.e. giving no RF). Here is what it prints out (it is the same as before):

`print:4`
`print:['0xffffffff'] ` (16 times)
`print:['0x40']`

jordens, 2020-10-13:
Could you post the system description json, the device db and the exact example script?

jordens, 2020-10-13:
I don't see anything obvious.

* Completely remove or comment that `f.set_cfg(dac_sleep=0, trf0_ps=1, trf1_ps=1)` line.
* Could you try a bitstream with just the standard json description (`python -m artiq.gateware.targets.kasli_generic phaser.json`): and `phaser.json`:

```json
{
    "target": "kasli",
    "hw_rev": "v2.0",
    "variant": "phaser",
    "base": "standalone",
    "core_addr": "10.34.16.100",
    "peripherals": [
        {
            "type": "phaser",
            "ports": [
                0
            ]
        }
    ]
}
```

* If that doesn't get us anywhere I'll make a Kasli bitstream for you (this is Kasli v1.1, right?)

sotirova, 2020-10-15:
> Completely remove or comment that `f.set_cfg(dac_sleep=0, trf0_ps=1, trf1_ps=1)` line.

Didn't make a difference

> Could you try a bitstream with just the standard json description (python -m artiq.gateware.targets.kasli_generic phaser.json): and phaser.json

Just tried this as well, with the line above line still commented and this [bitstream](https://github.com/quartiq/phaser/releases/download/v0.4/phaser_v0.4.bit) but have seen no RF yet

> If that doesn't get us anywhere I'll make a Kasli bitstream for you (this is Kasli v1.1, right?)

Yes, this is kasli v1.1

jordens, 2020-10-15:
Try these:
[phaser_kasli_v11_eem1.tar.gz](https://github.com/quartiq/phaser/files/5386034/phaser_kasli_v11_eem1.tar.gz)
artiq [`eecd97ce`](https://github.com/m-labs/artiq/tree/eecd97ce)
misoc [`a8a33f65`](https://github.com/m-labs/misoc/tree/a8a33f65)
phaser 92bca29
Vivado 2020.1

I've tested this with Kasli v2.0, Phaser in EEM1, and clk_sel=1 (external SMA, see device_db) and both a Phaser-Upconverter and a Phaser-Baseband.

It's also not impossible that this is a hardware issue. Maybe try https://github.com/elhep/phaser_gw as well.

sotirova, 2020-10-23:
With these kasli/phaser bitstreams I have finally managed to get RF out of both phaser boards we have, thank you very much for the help! However, the interpolation filter initialization is still failing. I am using the artiq/misoc/migen/Vivado versions as per your last comment. I have connected Phaser EEM0 to Kasli EEM1 and INT CLK IN of phaser to J3 on Kasli (have adjusted the device_db accordingly).

jordens, 2020-10-23:
* I assume with "interpolation filter failing" you mean you get noise out of the DACs if you bypass the "DUC/oscillator phase/amplitude test". Please confirm.
* Try re-loading the bitstream until you get the right test signal out of it. Give up if it it still doesn't work after the 10th time.
* Please try figuring out why your own bitstreams don't work.

RHanley1, 2020-11-26:
Hi Robert, I have taken over the testing of phaser from Ana. 

Using the kasli/phaser bitstreams you sent, and creating a conda environment with the suggested package git hashes above, I have managed to get some RF out of phaser also. However this seems to be volatile when reloading kasli. I have **not** commented out the DUC/oscillator test in the phaser init function, and this seems to pass without issue. The testing script I have been using is 
```
from artiq.experiment import *

class PhaserRF(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("phaser0")

    @kernel
    def run(self):
        f = self.phaser0
        self.core.reset()
        self.core.break_realtime()
        f.init()
        self.core.break_realtime()
        for ch in range(2):
            f.channel[ch].set_att(0*dB)
            f.channel[ch].set_duc_frequency(100*MHz)
            f.channel[ch].set_duc_phase(0.)
            f.channel[ch].set_duc_cfg(select=0, clr=0)
            delay(.1*ms)
            for osc in range(5):
                if osc == 4:
                    asf = 0.2
                else:
                    asf = 0.0
                freq = 0*MHz
                f.channel[ch].oscillator[osc].set_frequency(freq)
                delay(0.1*ms)
                f.channel[ch].oscillator[osc].set_amplitude_phase(asf, phase=0., clr=0)
                delay(0.1*ms)
        print("Done")
```

  Below is an example of what I have seen during testing:

Note             |  Plot
:-------------------------:|:-------------------------:
Run ``` artiq_flash start ```, then the above script | [image]
Run the above script again | [image]
Run the above script with asf=0 for all channnels | [image]
Run ``` artiq_flash start ```| - 
Run the above script with asf=0 for all channnels | [image]
Run the above script | [image]

jordens, 2020-11-26:
Yes. If you reload/restart/cycle/flash etc Kasli, you'll need to reload the bitstream to Phaser to reinitialize it cleanly and then run your experiment.
Fixing this issue here will remove the need to reload the Phaser bitstream but you'll still have to re-initialize Phaser and re-run your experiments.

RHanley1, 2020-11-26:
Is this still the case if one flashes the bitsream rather than just loading it?

jordens, 2020-11-26:
The work around is resetting the FPGA on Phaser. That's done either by flashing or loading. Loading is obviously less stressfull on the flash memory and much faster.

hartytp, 2020-11-30:
@jordens this is quite annoying to work around. What's the plan/timeline for resolving it?

jordens, 2020-11-30:
Unless you are developing on the gateware this should only hit you less than once per system reboot. That seems acceptable for the characterization phase.
It'll be resolved as soon as I find some time and I'm happy that your tests and characterization don't uncover major other issues that would require bigger rework.

hartytp, 2020-11-30:
@jordens currently we're seeing quit a bit of non-reproducible behaviour. We have also been swapping Kasli bitstreams around (both debugging issues with why our bitstreams weren't working and changing clocking options on Kasli to debug noise issues). Having to get into a dance of restarting Kasli and reloading phaser until things behave as expected is quite a time sink. It also means that if things happen that we don't expect, it's hard to understand if we're doing something silly or if phaser just didn't initialise correctly without going through a few reset cycles.

Annoyingly vivado is playing up and refusing to recognise the hardware on the computer we're using. This is not directly an issue with phaser, but it is annoying that we need to have a programmer hooked up to allow us to do the reloading dance even though we're not changing phaser bit streams. Having the I2C tool working would at least make this process easier at our end (we're using a kasli 1.1 so no 2.0 support needed atm).

hartytp, 2020-11-30:
@RHanley1 should comment, but I believe that the specific issue he's seeing is that he's getting spurious output when using 3 tones on phaser. This had previously been working so it's likely that phaser just needs to be reloaded, but vivado is now playing up.

RHanley1, 2020-11-30:
Current situation:

| Number of tones | Plot |
| ------------- | ------------- |
| Single tone at 0MHz | [image]  |
| Two tones, +/- 5MHz | [image] |
| Three tones, 0MHz and +/- 5MHz | [image]  |

jordens, 2020-11-30:
On I2C flashing: as I explained, the code is there, you need to try it. It will be very slow. Don't magically expect to be comfortable. I don't have a Kasli-v1.1 to test it with.

On the Phaser resets, please re-read my explanations. There is no need to do a dance with reloading Kasli to work around this issue. Let's first fix the easy silly issues and then see whether the remaining are blockers.

Vivado can leave a `hw_server` and a `cs_server` around if it is unhappy about something. That'll block.

jordens, 2020-11-30:
@RHanley1 please stop posting unrelated stuff on this issue.

hartytp, 2020-11-30:
@RHanley1 can you open a new issue for that? https://github.com/quartiq/phaser/issues/5#issuecomment-735813634

closed 2020-12-08 by jordens

SingularitySurfer, 2020-12-08:
Hey, I implemented some other interpolation filtes in https://github.com/quartiq/phaser/commit/b36e506b08382969e785597de0cc0e6c222b0445 which should fix the instability for now.

commit b36e506b (2020-12-08): Merge pull request #13 from quartiq/instability_fix

Instability fix

Close #5 

The goal should be to fix the issue in the misoc filter (due to stuttering of the input data while the link is being established leading to different beats and variation in the beats between the downstream HBFs/CICs ACK and the upstream frame STB). Once that's fixed revert to the misoc filter because it uses fewer resources.