quartiq/phaser #10: dac fifo alarms after cold boot

jordens, 2020-09-17:
The DAC signals all possible FIFO alarms after cold boot, reset and init sequence. Doing reset and init sequence again clears them.

kaolpr, 2020-09-17:
I was trying to set up Phaser with latest codes from this repo for @gkasprow, but it fails with "DAC Alarm" and alarm code 2112, what corresponds to `alarm_from_iotest` and `alarms_from_fiflo` = 0x2. Is this what you've observed?

What do you mean by reset in this case?

jordens, 2020-09-17:
The iotest is just the way I do the test (will disappear in a commit that I have pending). If the fifo error is persistent then you need to tweak the fifo_offset (see the comment in the code).
Reset is the DAC reset.

But let me sort the code and tag what you should be using.

jordens, 2020-09-17:
Use the following to play with it:

misoc 169dd32c
artiq f0959fb8
phaser cd2a893

jordens, 2020-09-22:
fifo alignment was too soon after pll setup and settling. fixed in init sequence.

closed 2020-09-22 by jordens

gkasprow, 2020-09-24:
I still have this issue.

gkasprow, 2020-09-24:
I updated misoc, qrtiq, phaser, compiled, flashed KASLI, programmed Phaser  and still have some issue

`
artiq_run ./experiments/phaser_playground.py 
2112
Core Device Traceback (most recent call last):
  File "./experiments/phaser_playground.py", line 12, in __modinit__ (RA=+0x784)
    self.phaser0.init()
  File "<artiq>/coredevice/phaser.py", line 247, in ... artiq.coredevice.phaser.Phaser.init<artiq.coredevice.phaser.Phaser>(...) (RA=+0x1f68)
    self.check_dac_alarms()
  File "<artiq>/coredevice/phaser.py", line 255, in ... artiq.coredevice.phaser.Phaser.check_dac_alarms<artiq.coredevice.phaser.Phaser>(...) (RA=+0x2c60)
    raise ValueError("DAC alarm")
  File "<artiq>/coredevice/phaser.py", line 255, column 13, in ... artiq.coredevice.phaser.Phaser.check_dac_alarms<artiq.coredevice.phaser.Phaser>(...)
    raise ValueError("DAC alarm")
    ^
ValueError(0): DAC alarm
Traceback (most recent call last):
  File "/nix/store/ks8m3p0isn9l87795ap258l4abc5p9g9-python3.7-artiq-6.7267.4340a5cf.beta/bin/.artiq_run-wrapped", line 9, in <module>
    sys.exit(main())
  File "/modules/artiq/artiq/frontend/artiq_run.py", line 225, in main
    return run(with_file=True)
  File "/modules/artiq/artiq/frontend/artiq_run.py", line 211, in run
    raise exn
  File "/modules/artiq/artiq/frontend/artiq_run.py", line 204, in run
    exp_inst.run()
  File "/modules/artiq/artiq/language/core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/modules/artiq/artiq/coredevice/core.py", line 137, in run
    self.comm.serve(embedding_map, symbolizer, demangler)
  File "/modules/artiq/artiq/coredevice/comm_kernel.py", line 516, in serve
    self._serve_exception(embedding_map, symbolizer, demangler)
  File "/modules/artiq/artiq/coredevice/comm_kernel.py", line 508, in _serve_exception
    raise python_exn
ValueError: DAC alarm
`

jordens, 2020-09-24:
This is using the 125 mhz from kasli?

jordens, 2020-09-24:
This issue covers the case where running the same script again (init() again) will not result in an error. Does it?