quartiq/phaser #11: close-in noise

RHanley1, 2020-11-27:
I am currently testing a Phaser baseband board using the kasli and phaser bitstreams provided in #5. I am currently just producing a single tone at 100MHz. The script for producing the tones is below:
```
from artiq.experiment import *

class PhaserRF(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("phaser0")

    @kernel
    def run(self):
        f = self.phaser0
        self.core.reset()
        self.core.break_realtime()
        f.init(debug=True)
        self.core.break_realtime()
        for ch in range(2):
            f.channel[ch].set_att(0*dB)
            f.channel[ch].set_duc_frequency(100*MHz)
            f.channel[ch].set_duc_phase(0.)
            f.channel[ch].set_duc_cfg(select=0, clr=0)
            delay(.1*ms)
            freq = 0*MHz
            asf = 0.0
            for osc in range(5):
                if osc == 4:
                    asf = 0.5
                    freq = 0*MHz
                elif osc == 3:
                    asf = 0.0
                    freq = 0*MHz
                elif osc == 2:
                    asf = 0.0
                    freq = 0*MHz
                elif osc == 1:
                    asf = 0.0
                    freq = 0*MHz
                elif osc == 0:
                    asf = 0.0
                    freq = 0*MHz
                f.channel[ch].oscillator[osc].set_frequency(freq)
                delay(0.1*ms)
                f.channel[ch].oscillator[osc].set_amplitude_phase(asf, phase=0., clr=0)
                delay(0.1*ms)
        f.duc_stb()
        print("Done")
```

I get the output as expected (see below), however, if I take a closer look at the spectrum around 100MHz, there appears to be a lot of close-in noise. Is this something you have observed, or am I doing something silly? 
| Wide scan  | Close scan |
| ------------- | ------------- |
| [image] | [image] |

hartytp, 2020-11-27:
@RHanley1 thanks for posting the data. AFAICT this is more a HW thing than a SW/GW thing, so the issue belongs over at https://github.com/sinara-hw/phaser/issues

hartytp, 2020-11-27:
@RHanley1 did you check the quality of the 125MHz clock from Kasli to make sure it's clean?

jordens, 2020-11-27:
Thanks for exploring Phaser.
But can we try to use the issue tracker on the phaser gateware repository for its development and not so much for lab notes, support questions, etc? We have much better platforms (other repositories, the forum, chats etc) for that.
Once you are reasonably convinced that you have identified a gateware issue on Phaser, please do post that here (comment here if this turns out to be a gateware issue). Hardware issues and hardware characterization should go the hardware repository.

closed 2020-11-27 by jordens