quartiq/phaser #16: DSP Block Diagram

RHanley1, 2021-03-16:
It would be useful if there was a functional block diagram of the DSP to give a nice overview of phaser, given its non-trivial nature. There is an old diagram here (sinara-hw/Phaser#1) , however I believe this is not quite up-to-date with the latest release. Does an up-to-date diagram exist?

jordens, 2021-03-16:
That one is still fundamentally correct for phaser-classic

RHanley1, 2021-03-19:
Assuming I have interpreted the description in the core-device driver correctly, the “Phaser Oscillators” are the blocks in the “Kasli RTIO Gateware” (there are only 4 on the diagram but 5 in reality?), there is a DUC on the FPGA which one sets with “set_duc_frequency” etc. (which isn’t in this diagram), which then can be mixed with the NCO which one sets with “set_nco_frequency” in the DAC (which is in the diagram)? Is this correct? If so, could you update the diagram accordingly, as a pictorial representation makes is significantly easier to interpret the functions in the driver?

SingularitySurfer, 2021-03-19:
Hey, yes there is one more DUC in the Phaser FPGA not shown in this diagram. Your interpretation of the driver is right.

jordens, 2021-03-19:
@RHanley1 This is open source. If you have the need for extended or updated documentation, or other additional support, development, or work on phaser, please feel free to get in touch. Otherwise high-quality contributions are always welcome.

SingularitySurfer, 2021-03-29:
There's an updated block diagram and a bit of high-level DSP description in the [wiki](https://github.com/quartiq/phaser/wiki) now. Also some documentation for the STFT pulsegen. Please feel free to give suggestions.

closed 2021-03-29 by jordens

hartytp, 2021-03-29:
That's really useful! Thanks

RHanley1, 2021-03-30:
Thanks very much @SingularitySurfer! That makes the core-device driver much easier to understand