quartiq/phaser #17: Wrong values from trf_read

Jacky2k, 2021-04-07:
Hi everyone,
I am trying to program the TRF PLL/VCO/mixer on the Phaser and I am having some issues when reading back the registers. For example:
`self.phaser0.channel[0].trf_write(0x08A0320A)
self.phser0.channel[0].trf_read(2)`
Yields 0xFFA0320A. There are other registers where this is also a problem. Some leading bytes are always read back as 0xFF. I could not find a hint in the datasheet that this is intended.
Additionally, it is not clear if I should call trf_read(2) or trf_read(10) if I want to read register 2. According to the datasheet register 2 has the address 10 and in your example code you are printing from trf_read(0) to trf_read(7) while calling the variable addr, this is very confusing.

jordens, 2021-04-07:
The datasheet doesn't say that this is not intended either.
The documentation clearly states that the register address to read is 0-7.
https://github.com/m-labs/artiq/blob/78490bef5d894e610bf4555f5b03071dc2115810/artiq/coredevice/phaser.py#L874-L880
But you can also use `2 | 8 = 10` if you want. It shouldn't change the outcome.
What is the actual issue here?

Jacky2k, 2021-04-07:
Well, I don't know what the actual issue is. That is why I was reading back the registers to make sure they are what I set them to. Once I manually set the registers I get no more output from the Phaser, although the LO output shows the frequency I intended. Additionally, the LO output on the MMCX is very low (like -20 dBm), which seems strange. Also the output of the Phaser when not touching the trf registers is very low. A full amplitude sine-wave generates only -10 dBm. Similar results for the supplied example.py.
I am trying to figure out what is wrong. What is the PLL initialized to after reset? I know where to find the reset register values, but translating them back to a frequency with a very high certainty is not trivial.

jordens, 2021-04-07:
The user needs to investigate and tune PLL parameters depending on the application. There is no one-size-fits-all. The current default ones are mostly just the ones [Greg found to simply work](https://github.com/sinara-hw/Phaser/issues/101#issuecomment-653691964).
Also this is the repository for the phaser gateware. The ARTIQ driver and the code you refer to is over at https://github.com/m-labs/artiq. There are a bunch of issues open. https://github.com/m-labs/artiq/issues?q=is%3Aissue+is%3Aopen+label%3Aarea%3Aphaser
Closing as invalid as it likely has nothing to do with the gateware.

closed 2021-04-07 by jordens

Jacky2k, 2021-04-07:
I don't get why reading registers has nothing to do with the gateware

jordens, 2021-04-07:
Reading registers indeed also involves gateware. If you want to insist that there is an issue in the gateware, fine, but it's unlikely to help in any way. Please do take the advice and look at the issues I linked for you.
There is no evidence that the issue you report (not being able to configure the PLL) has anything to do with the phaser gateware. The code you reference is not the phaser gateware. The answer to the question you ask is not here.