quartiq/phaser #23: FastLink IDELAY2.{ld,ce} not hooked up.

shareefj, 2023-12-22:
I've been looking at the FastLink implementation to better understand what's going on and noticed that neither the CE or LD pins of the IDELAYE2 primitive are hooked up.  There seems to be logic to modify the tap count but this will never be loaded.  Is this an oversight or just historical code still lurking?

The LD signal is defined [here](https://github.com/quartiq/phaser/blob/master/link.py#L10) but is never assigned to.  Checking the Verilog netlist and both nets are driven to 1'b0.

jordens, 2023-12-22:
What's the issue?

shareefj, 2023-12-28:
Sorry, I should have provided some context.

I'm in the process of trying to understand and modify Phaser for our own use and haven't found any form of implementation documentation and the code itself seems to be shy of comments.  So I'm spending some time reverse engineering it in order to document the major interfaces such as Fastlink.

So I guess my first question should have been is there any documentation for Fastlink anywhere?

Secondly, my original question came out of looking in detail at the interface.  To me it looks like the IDELAY elements aren't used which could mean it's a bug.  Or it was copy-pasted from another module and you're assuming the synthesis tool will optimise it.  Or my understanding of the block itself is flawed.  The Xilinx docs seem to say that in "VAR_LOAD" mode either the CE or LD inputs must be asserted to modify the tap value.  This isn't the case here.

So for my understanding of the interface, can you either point out my error or explain why the ports haven't been connected?  Thanks.

jordens, 2023-12-28:
I currently don't have the bandwidth to provide coding support. There is no further documentation than what's available in the repositories. Quality PRs with documentation or new developments are certainly welcome.

jordens, 2023-12-30:
The delays are just fixed. Not a bug, not a feature.

closed 2023-12-30 by jordens