quartiq/phaser #24: output glitches at ASF=1.0

nkrackow, 2025-04-28:
## HW Setup
Phaser Baseband V1.1.1 connected to Kasli SoC.

## Phaser setup
- current ARTIQ 7 
- single oscillator at 1 MHz
- DUC at 100 MHz
- attenuator at 0 dB

## Problem

Setting the ASF to 0.9999 using `.set_amplitude_phase()` yields a rock-solid sinewave without glitches or anything for minutes. I was never able to trigger on anything or saw anything in the scope persistence mode.

Setting the output to 0.99999 (which rounds to 1.0) or 1.0 does not yield "ValueError: amplitude out of bounds" (1.0001 does) in ARTIQ but leads to very sporadic bursts of output glitches. The glitch frequency within a burst varies but is in the hundreds of us range. The bursts happen irregularly every few seconds.

My best guess would be that the interpolators have a chance to produce a sample that overflows close or at the maximum ASF.

[image]

[image]

[image]

jordens, 2025-04-28:
There is a warning about that:

https://github.com/m-labs/artiq/blob/8156151589019dd13770272e2f575234336725ea/artiq/coredevice/phaser.py#L869-L872

But that's not the full story. We both discussed that at length...

You need to make sure that the output does not clip (1) over all oscillators (2) including DC oscillators in the IQ square (3) over all your possible AC responses of the filter and (4) including intermediate filter stages and (5) including all transients.

May well be that there is an overflow under some conditions. You don't provide the MWE you use...

Simulate it!

Note that the docs and test in set_amplitude_phase are likely correct. They don't and can't concern themselves with the filters which come later (after the summation and on phaser).
There is no easy way to predict overflow conditions here. We want to support the full IQ square (i.e. DC components). One bit of headroom in the filters and saturation would still be an bad solution: you'd also have to consider the FIR overshoot which you never want to saturate away and which needs to survive all the way up to the output. That leaves you with sacrificing a Gibbs-sized headroom **in the DAC output range**. And to implement that it's easiest to do it at the start: just limit your amplitude sum to 0.9. Even for perfect filters, it's relatively easy to construct the worst case input signal that will need all the guard amplitude given the taps in the notebook.
The docs can always be expanded and made more accurate: PRs welcome.

nkrackow, 2025-04-29:
Thanks, I am aware that you have to you have to manage amplitudes with multiple tones, I just didn't think of this when using just one tone. I think this could easily happen to other users and the failure mode is kinda nasty (rare glitches every few seconds) which is why I made the issue. 
Even though it already clearly states that the sum has to be below 0 in the docs I [added the note about a single oscillator](https://github.com/m-labs/artiq/pull/2740).

jordens, 2025-04-29:
You misunderstand. I think there may well be a bug. The filters were designed to have headroom (IIRC CICCOMP has > 1 gain but 1 bit output headroom. FIRs have ~1 gain and 1 bit headroom and CIC has < 1 gain). The overall response is strictly less than 1 for precisely that reason (at least on paper). Hence for any single frequency, single oscillator, it should not overflow after all transients have settled. Can you debug?

nkrackow, 2025-04-29:
Yes, I think can invest some time in the coming days.

jordens, 2025-05-06:
#25

nkrackow, 2025-05-06:
ah, no this is actually unrelated to the MR..

nkrackow, 2025-05-06:
Ok, so it looks like the CIC compensator makes the whole interpolator frequency response just barely peak above 1 in the region between 1 and 1.4 MHz. I was never able to see the problem anywhere else on HW. I tried to check for this in simulation [here](https://github.com/nkrackow/phaser/blob/nk/debug2/test_interpolate.py#L59) but somehow I could never trigger it there. Maybe its just too seldom for simulation. 

Adding a second identical oscillator with amplitude 0.0001 leads to overflows all the time on HW. So does a sine with `1 << 13` amplitude in simulation. 

I can offer [this](https://github.com/quartiq/phaser/pull/26) but I don't have time to whip out the filter design tools again right now. With this I was never able to see glitches on HW.

nkrackow, 2025-05-06:
Forgot to mention that I validated overflows at the DAC input by latching the odd sample into a Phaser register if the even sample was at the rail (0x7fff) and sometimes I could see a sample below 0 (>=0x8000 in unsigned) aka an overflow happened.